victronenergy/venus #1403: Add deviceinstance field to notifications API

ReinvdZee, 2024-12-19:
The deviceinstance of the service triggering an alarm should be added to the [notifications API](https://github.com/victronenergy/venus/wiki/dbus#platform): `com.victronenergy.platform/Notifications/[0-19]/DeviceInstance`. 

This field is to be used as ID by the N2K alarm out function.

mpvader, 2025-04-16:
@ReinvdZee still v3.60?

ReinvdZee, 2025-04-16:
Mark bath (somehow can't tag him here) added the `/DeviceInstance` to venus-platform [here](https://github.com/victronenergy/venus-platform/commit/27ffd2bf320425d2d499d9581dcc25a1ac41c479). I'm not sure if this is all he needs for the N2K alarm out function. I've asked him on slack.

ReinvdZee, 2025-04-28:
@nmbath 

>If you Ould like me to rebase to latest code and re-test please let me know.

Yes, please rebase current master on your branch and retest.

ReinvdZee, 2025-05-08:
Merged and on the todo.

closed 2025-05-09 by jhofstee