victronenergy/venus #1406: Unsecure default settings FlashMQ / nginx on Cerbo GX

helgegrimm, 2024-12-24:
Device: Cerbo GX
Firmware: v3.51
Services: TLS enabled Nginx at Port 443 and FlashMQ on Port 8883

Cerbo GX was scanned via Greenbone CE. Results see below:

**1. DoS vulnerabilities:**
The remote SSL/TLS service is prone to a denial of service (DoS) vulnerability.
CVE-2011-1473 & CVE-2011-5094

The following indicates that the remote SSL/TLS service is affected:
Protocol Version | Successful re-done SSL/TLS handshakes (Renegotiation) over an existing / already established SSL/TLS connection
TLSv1.1          | 10
TLSv1.2          | 10

The flaw exists because the remote SSL/TLS service does not properly restrict client-initiated renegotiation within the SSL and TLS protocols.

**2. deprecated TLSv1.0 and 1.1 protocol usage**
It was possible to detect the usage of the deprecated TLSv1.0 and/or TLSv1.1 protocol on this system.

In addition to TLSv1.2+ the service is also providing the deprecated TLSv1.0 and TLSv1.1 protocols and supports one or more ciphers. Those supported ciphers can be found in the 'SSL/TLS: Report Supported Cipher Suites' (OID: 1.3.6.1.4.1.25623.1.0.802067) VT.

The TLSv1.0 and TLSv1.1 protocols contain known cryptographic   flaws like:
  - CVE-2011-3389: Browser Exploit Against SSL/TLS (BEAST)
  - CVE-2015-0204: Factoring Attack on RSA-EXPORT Keys Padding Oracle On Downgraded Legacy Encryption (FREAK)

An attacker might be able to use the known cryptographic flaws to eavesdrop the connection between clients and the service to get access to sensitive data transferred within the secured connection.

It is recommended to disable the deprecated TLSv1.0 and/or TLSv1.1 protocols in favor of the TLSv1.2+ protocols.

Possible solution nginx:
In /etc/nginx/nginx.conf disable TLSv1.0 and TLSv1.1 protocols by changing
ssl_protocols TLSv1 TLSv1.1 TLSv1.2 TLSv1.3; # Dropping SSLv3, ref: POODLE
into
ssl_protocols TLSv1.2 TLSv1.3; # Dropping SSLv3, ref: POODLE, dropping TLSv1 and TLSv1.1, ref: BEAST, FREAK

Possible solution FlashMQ:
build fixed version of FlashMQ, see https://github.com/halfgaar/FlashMQ/blob/master/bridgeconfig.cpp
        SSL_CTX_set_options(sslctx->get(), SSL_OP_NO_SSLv3); // TODO: config option
        SSL_CTX_set_options(sslctx->get(), SSL_OP_NO_TLSv1); // TODO: config option
add
        SSL_CTX_set_options(sslctx->get(), SSL_OP_NO_TLSv1_1); // disabling TLSv1_1

helgegrimm, 2024-12-25:
addendum possible solution FlashMQ:
also https://github.com/halfgaar/FlashMQ/blob/master/listener.cpp need to be extended with this line at the suitable position:
SSL_CTX_set_options(sslctx->get(), SSL_OP_NO_TLSv1_1); // disabling TLSv1_1

halfgaar, 2025-02-08:
Tracking FlashMQ's changes here: https://github.com/halfgaar/FlashMQ/issues/136

As for Venus using OpenSSL 1.1, that's beyond FlashMQ's influence. FlashMQ works with OpenSSL 3, and does so for the builds provided for modern Linux distros.

halfgaar, 2025-02-09:
FlashMQ now has support for `minimum_tls_version` in bridges and listeners. It will be part of 1.20, which will probably be released soonish.

jhofstee, 2025-02-21:
@helgegrimm can you explain please how we are vulnerable if at all, so we can exploit it. The first two seems to be disputed, the 2.1 seems rather browser orientated, which we don't have. So while we will be updating the min versions, can you provide test cases for it or are you just reporting scanner output?

helgegrimm, 2025-02-22:
Device: Cerbo GX
Firmware: v3.53 (has been updated meanwhile)

Yes, I was refering to scanner reports as I mentioned in my first post: the results were obtained by using Greenbone CE.

### Regarding TLSv1.0 and 1.1 protocol usage:
TLS 1.0 and TLS 1.1 were declared unsecure already in 2021 by IETF - see RFC 8996.
See https://datatracker.ietf.org/doc/html/rfc8996

After I read halfgaar's response:
VenusOS is using OpenSSL version 1.1.1w.
OpenSSL 1.1.1 series has reached its End of Life (EOL) September 2023.
This means: no more publicly available security fixes unless you have extended support.
See https://openssl-library.org/post/2023-09-11-eol-111/
**An upgrade to OpenSSL 3.x should be considered.**

### Regarding to vulnerabilities:
I recommend to use testssl.sh:
https://testssl.sh/ 

I took a deeper look with testssl.sh on the Ports 443 and 8883 of my Cerbo GX:
I will post only not-OK and deprecated findings.

**Results for port 443 (nginx):**
Testing protocols:
-  TLS 1      offered (deprecated)
-  TLS 1.1    offered (deprecated)

Testing cipher categories:

-  Obsolete CBC ciphers (AES, ARIA etc.)         offered

Testing vulnerabilities:

-  BEAST (CVE-2011-3389)                     TLS1: ECDHE-RSA-AES256-SHA ECDHE-RSA-AES128-SHA AES256-SHA CAMELLIA256-SHA AES128-SHA CAMELLIA128-SHA
                                            VULNERABLE -- but also supports higher protocols  TLSv1.1 TLSv1.2 (likely mitigated)
-  LUCKY13 (CVE-2013-0169), experimental     potentially VULNERABLE, uses cipher block chaining (CBC) ciphers with TLS. Check patches
 

**Results for port 8883 (FlashMQ):**
Testing protocols

-  TLS 1.1    offered (deprecated)

Testing cipher categories

-  Obsolete CBC ciphers (AES, ARIA etc.)         offered

Testing server preferences

-  Has server cipher order?     no (NOT ok)

Testing vulnerabilities

-  Secure Client-Initiated Renegotiation     VULNERABLE (NOT ok), potential DoS threat
-  LUCKY13 (CVE-2013-0169), experimental     potentially VULNERABLE, uses cipher block chaining (CBC) ciphers with TLS. Check patches

### Regarding test cases:
For BEAST or LUCKY13 attacks you will need a artifical setup with you in the MITM position (in the same WiFi or LAN using ARP spoofing).
Additionally you need an old browser like Firefox 4, Chrome 21 or IE6/IE8 without updates _OR_ an old OpenSSL version on the client.
You can then use Metasploit for testing (they have modules for both attacks).

### Conclusion:
I have to admit: The attacks are not easy to exploit if current browsers are used. Nevertheless, security by design and hardening should be taken into account in order to avoid any doubts about a secure product.

closed 2025-02-22 by helgegrimm