victronenergy/venus #1409: Build Issue

charlesrg, 2025-01-06:
While following the build process **make fetch** breaks with:
```
--------------- sources/meta-noto - git@github.com:victronenergy/meta-noto.git -------------
Cloning into 'sources/meta-noto'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: not a git repository: 'sources/meta-noto/.git'
fatal: not a git repository: 'sources/meta-noto/.git'
fatal: not a git repository: 'sources/meta-noto/.git'
fatal: not a git repository: 'sources/meta-noto/.git'
```

Followed process as described:
_# clone this repository
git clone https://github.com/victronenergy/venus.git
cd venus

# install host packages (Debian based)
sudo make prereq

# fetch needed subtrees
# use make fetch-all instead, if you have access to all the private repos.
make fetch_

darkcheyenne, 2025-02-07:
I have the same issure following this guide:

https://github.com/victronenergy/venus

jhofstee, 2025-02-07:
right, perhaps you don't have a valid github login?

jhofstee, 2025-02-10:
For what it is worth, I deleted downloading meta-noto, see https://github.com/victronenergy/venus/commit/a72e8c96e213810f4ee0aac84adc1acb8c4537e8, it is no longer used anyway. That said, toying around with Venus OS is typically a developer thing, don't be surprised that you need a github account. It is not strictly needed, but most developers will never see these issues.

closed 2025-02-10 by jhofstee