victronenergy/venus #1410: Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8

twitnic, 2025-01-09:
Hello,
I hope I've come to the right place.
I have a Raspberry Pi CM4 freshly installed with VenusOS. Also activated the USB ports.

Connected an EM540 via RS485.
But this is not visible in VenusOS.

Here a Logfile:

root@raspberrypi4:~# cat /data/log/dbus-cgwacs.ttyUSB0/current 
@4000000067800e4038bb5a84 INFO  dbus-cgwacs v2.0.23 started 
@4000000067800e4038bb6e0c INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800e4038bb79c4 INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800e4038bb857c INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800e4038f031dc INFO  Wait for local settings on DBus...  
@4000000067800e4039572d24 INFO  Local settings found 
@4000000067800e4403faf19c qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
@4000000067800e7105188ecc *** starting dbus-cgwacs ***
@4000000067800e71076885c4 Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@4000000067800e710768994c Qt depends on a UTF-8 locale, but has failed to switch to one.
@4000000067800e710768a8ec If this causes problems, reconfigure your locale. See the locale(1) manual
@4000000067800e710768b88c for more information.
@4000000067800e7107753f94 INFO  dbus-cgwacs v2.0.23 started 
@4000000067800e7107779d0c INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800e7107799cc4 INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800e71077e2cbc INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800e7107e7375c INFO  Wait for local settings on DBus...  
@4000000067800e71081ce58c INFO  Local settings found 
@4000000067800e740405a3e4 qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
@4000000067800e9f0d93d28c *** starting dbus-cgwacs ***
@4000000067800e9f0fe4235c Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@4000000067800e9f0fe43eb4 Qt depends on a UTF-8 locale, but has failed to switch to one.
@4000000067800e9f0fe44a6c If this causes problems, reconfigure your locale. See the locale(1) manual
@4000000067800e9f0fe45a0c for more information.
@4000000067800e9f0fee4904 INFO  dbus-cgwacs v2.0.23 started 
@4000000067800e9f0fef46ec INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800e9f0fef52a4 INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800e9f0ff26f84 INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800e9f1057960c INFO  Wait for local settings on DBus...  
@4000000067800e9f108bcd3c INFO  Local settings found 
@4000000067800ea212e7419c qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
@4000000067800ecf1741fcb4 *** starting dbus-cgwacs ***
@4000000067800ecf19d03cfc Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@4000000067800ecf19d05084 Qt depends on a UTF-8 locale, but has failed to switch to one.
@4000000067800ecf19d06024 If this causes problems, reconfigure your locale. See the locale(1) manual
@4000000067800ecf19d06bdc for more information.
@4000000067800ecf19d07794 INFO  dbus-cgwacs v2.0.23 started 
@4000000067800ecf19d0834c INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800ecf19d08b1c INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800ecf19d194bc INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800ecf1a42a224 INFO  Wait for local settings on DBus...  
@4000000067800ecf1a9a43bc INFO  Local settings found 
@4000000067800ed221dd3a44 qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
@4000000067800f05093ec0ac *** CCGX booted (0) ***
@4000000067800f3422aaa04c *** starting dbus-cgwacs ***
@4000000067800f34260e745c Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@4000000067800f34260e8bcc Qt depends on a UTF-8 locale, but has failed to switch to one.
@4000000067800f34260e9b6c If this causes problems, reconfigure your locale. See the locale(1) manual
@4000000067800f34260eab0c for more information.
@4000000067800f34261a630c INFO  dbus-cgwacs v2.0.23 started 
@4000000067800f34261b60f4 INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800f34261b6cac INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800f34261e4ef4 INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800f342690371c INFO  Wait for local settings on DBus...  
@4000000067800f3426c6fe8c INFO  Local settings found 
@4000000067800f3722362014 qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
@4000000067800f7b090adf6c *** starting dbus-cgwacs ***
@4000000067800f7b0be9f704 Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@4000000067800f7b0bea125c Qt depends on a UTF-8 locale, but has failed to switch to one.
@4000000067800f7b0bea21fc If this causes problems, reconfigure your locale. See the locale(1) manual
@4000000067800f7b0bea319c for more information.
@4000000067800f7b0bf4ef9c INFO  dbus-cgwacs v2.0.23 started 
@4000000067800f7b0bf5ed84 INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800f7b0bf5fd24 INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800f7b0bf925bc INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800f7b0c8f81ec INFO  Wait for local settings on DBus...  
@4000000067800f7b0cce57dc INFO  Local settings found 
@4000000067800f7e12bbcc9c qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
@4000000067800fab0edecd04 *** starting dbus-cgwacs ***
@4000000067800fab113f8cdc Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@4000000067800fab113fa834 Qt depends on a UTF-8 locale, but has failed to switch to one.
@4000000067800fab113fb7d4 If this causes problems, reconfigure your locale. See the locale(1) manual
@4000000067800fab113fc38c for more information.
@4000000067800fab114a3b3c INFO  dbus-cgwacs v2.0.23 started 
@4000000067800fab114b353c INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800fab114b44dc INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800fab114e6d74 INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800fab11b75cbc INFO  Wait for local settings on DBus...  
@4000000067800fab11e72b04 INFO  Local settings found 
@4000000067800fae12b7408c qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
@4000000067800fdb13e99284 *** starting dbus-cgwacs ***
@4000000067800fdb163c15d4 Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@4000000067800fdb163c295c Qt depends on a UTF-8 locale, but has failed to switch to one.
@4000000067800fdb163c38fc If this causes problems, reconfigure your locale. See the locale(1) manual
@4000000067800fdb163c44b4 for more information.
@4000000067800fdb1648950c INFO  dbus-cgwacs v2.0.23 started 
@4000000067800fdb164afe3c INFO  Built with Qt 6.6.3 running on 6.6.3 
@4000000067800fdb164cfdf4 INFO  Built on Jan  7 2025 at 15:50:27 
@4000000067800fdb16514f6c INFO  Connecting to "/dev/ttyUSB0" 
@4000000067800fdb16b6c7fc INFO  Wait for local settings on DBus...  
@4000000067800fdb16f0c3bc INFO  Local settings found 
@4000000067800fde127dc99c qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 

Has anyone Idea?
VenusOS 3.53

twitnic, 2025-01-09:
Tested with 3.52 same Problem.

And with the newest Beta 3.60-15, also the same:

==> /var/log/dbus-cgwacs.ttyUSB0/current <==
@40000000678042101886ec4c *** starting dbus-cgwacs ***
@40000000678042101afc60ec Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
@40000000678042101afc7c44 Qt depends on a UTF-8 locale, but has failed to switch to one.
@40000000678042101afc93b4 If this causes problems, reconfigure your locale. See the locale(1) manual
@40000000678042101afca73c for more information.
@40000000678042101b076924 INFO  dbus-cgwacs v2.0.24 started 
@40000000678042101b086324 INFO  Built with Qt 6.6.3 running on 6.6.3 
@40000000678042101b0872c4 INFO  Built on Jan  8 2025 at 13:15:26 
@40000000678042101b0be594 INFO  Connecting to "/dev/ttyUSB0" 
@40000000678042101b753e54 INFO  Wait for local settings on DBus...  
@40000000678042101ba4e974 INFO  Local settings found 
@40000000678042131d95b524 qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) :

twitnic, 2025-01-11:
I tried a Raspberry Pi 3 with a new Installation. The Same ...

------------------->%------------------
==> /var/log/dbus-cgwacs.ttyUSB0/current <==
@400000006782912c2f09f43c Qt depends on a UTF-8 locale, but has failed to switch to one.
@400000006782912c2f0a03dc If this causes problems, reconfigure your locale. See the locale(1) manual
@400000006782912c2f0a1764 for more information.
@400000006782912c2f182cdc INFO  dbus-cgwacs v2.0.23 started 
@400000006782912c2f1b8454 INFO  Built with Qt 6.6.3 running on 6.6.3 
@400000006782912c2f1d64cc INFO  Built on Jan  7 2025 at 15:50:27 
@400000006782912c2f21cdb4 INFO  Connecting to "/dev/ttyUSB0" 
@400000006782912c2faa00e4 INFO  Wait for local settings on DBus...  
@400000006782912c2fed478c INFO  Local settings found 
@400000006782912f2c169154 qt.dbus.integration: Could not disconnect "org.freedesktop.DBus" to onServiceOwnerChanged(QString,QString,QString) : 
------------------->%------------------

feldsoft, 2025-02-03:
Hi Twitnic,
I'm running a Raspi 4B with V3.51. EM24 with 485 / USB converter.

By checking the logs I saw the same like you saw in your logs.

/data/log/dbus-cgwacs.ttyUSB1/tmp

2025-02-03 15:04:43.203160500 *** starting dbus-cgwacs ***
2025-02-03 15:04:43.244995500 Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
2025-02-03 15:04:43.245002500 Qt depends on a UTF-8 locale, but has failed to switch to one.
2025-02-03 15:04:43.245005500 If this causes problems, reconfigure your locale. See the locale(1) manual
2025-02-03 15:04:43.245009500 for more information.
2025-02-03 15:04:43.245650500 INFO  dbus-cgwacs v2.0.22 started 
2025-02-03 15:04:43.245715500 INFO  Built with Qt 6.6.3 running on 6.6.3 
2025-02-03 15:04:43.245719500 INFO  Built on Nov  4 2024 at 18:35:44 
2025-02-03 15:04:43.245906500 INFO  Connecting to "/dev/ttyUSB1" 
2025-02-03 15:04:43.254449500 INFO  Wait for local settings on DBus...  
2025-02-03 15:04:43.257697500 INFO  Local settings found 
2025-02-03 15:04:43.308260500 INFO  Device ID: 71 
2025-02-03 15:04:44.203375500 INFO  Device found: "BX11600630014" @ "/dev/ttyUSB1" 
2025-02-03 15:04:46.748742500 INFO  Registering service "com.victronenergy.grid.cgwacs_ttyUSB1_mb1" 
2025-02-03 15:04:46.782104500 [VeQItemDbusPublisher] Registered service "com.victronenergy.grid.cgwacs_ttyUSB1_mb1"

I would also be interested to know what that means.

Maybe someone can check the log on a Cerbo?

Regards Alfred

izak, 2025-02-04:
Hi @twitnic , the message about the encoding is a bit annoying, but ultimately not the cause, it just means the logging is being done using plain ascii, which is fine.

Please check the baud rate you are using to communicate with the meter. This needs to be 9600 baud. The log file suggests that it cannot communicate with the meter. The text "Local settings found" is the last thing it prints before the meter is found, and "qt.dbus.integration: Could not disconnect ..." is the last thing that happens as the process terminates after it gives up. So that log indicates it cannot communicate with the meter. Since you reported that it works to a laptop, I think the baud rate is the most likely thing to check.

izak, 2025-02-04:
Hi @twitnic , thanks for the response. These devices ship with a default of 9600 baud and work out of the box. Or at least, that was always the case in the past. I will ask our documentation expert to add a note about this issue.

closed 2025-02-04 by izak

twitnic, 2025-02-04:
@izak 

> These devices ship with a default of 9600 baud 

FYI: It was a new Device from here: https://www.offgridtec.com/victron-energy-meter-em540-3-phasig-energiemessgeraet-max-65a-phase.html