victronenergy/venus #1412: Add I/O extender support to Venus

dirkjanfaber, 2025-01-14:
This issue is about adding support for the I/O extender (and other switching devices).

When plugged in the GX, the device adds digital inputs as we are used to, (under `com.victronenergy.digital_input` etc) because it is an USB device. The device does contain an eeprom, so we know the serial (see `dmesg` output below).

More complex devices end up under `com.victronenergy.switch.iox_$serial`, and perhaps some of the configuration of the io extender will be placed there too. E.g. the state of the dipswitches. 

Affected repositories (needs to be checked, possibly not altered):
- [ ] systemcalc ibu / rvz
- [ ] node-red-contrib-victron dfa
- [ ] venus-platform rvz
- [ ] gui jho
- [ ] gui-v2 rvz
- [ ] dbus_modbus_tcp dfa
- [ ] vecan-dbus mbo
- [ ] dbus-generator (running state, digital input) rvz

```
[511157.709507] usb 6-1: new full-speed USB device number 3 using ohci-platform
[511157.992646] usb 6-1: New USB device found, idVendor=10c4, idProduct=ea90, bcdDevice= 0.00
[511158.001143] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[511158.008408] usb 6-1: Product: GX IO-Extender 150
[511158.013229] usb 6-1: Manufacturer: Victron Energy BV
[511158.018430] usb 6-1: SerialNumber: HQ24200AFFE
[511158.032731] cp2112 0003:10C4:EA90.0002: hidraw0: USB HID v1.01 Device [Victron Energy BV GX IO-Extender 150] on usb-1c1d400.usb-1/input0
[511158.120561] cp2112 0003:10C4:EA90.0002: Part Number: 0x0C Device Version: 0x03
```

ReinvdZee, 2025-01-22:
On Venus OS 3.60~17, the IO extender is recognized properly. The inputs will be picked up by `dbus-digitalinputs`. The GUI used to show the digital inputs by deriving the input number from the path `com.victronenergy.settings/Settings/DigitalInput/<n>/` to show it as "Digital Input <n>". This only accepted integers. I changed the GUI to show the input label instead: `com.victronenergy.digitalinputs/Devices/x/Label`, like the GUI does for analog inputs.

Now, the digital inputs are automatically picked up and shown as:

[image]

mpvader, 2025-08-19:
this was all done - closing.