victronenergy/venus #1418: Detect generator at DC input alarm

ReinvdZee, 2025-01-30:
For AC generators, we provide the feature to trigger an alarm when there is no AC detected on the generator input after a period of time (5 minutes):

[image]

We should provide a similar feature for DC gensets. Also, hide the AC detection feature when there is a DC genset connected and vice-versa.

For a DC genset, use the current reading of the genset for the detection and trigger an alarm if the current has been below 5A during the whole detection window. The setting is disabled by default and will not be added to GUI-V1.

Projects involved:
- [ ] dbus-generator
- [ ] gui-v2
- [ ] venus-platform (alarm)

ReinvdZee, 2025-01-31:
I added a new alarm setting, which is only visible for DC genset services:

[image]

The setting is never visible for the relay controlled genset. The "Detect generator at AC input" setting is hidden in case a DC genset is connected.

When the setting is enabled, an alarm will be triggered if the DC current read by the genset has not been above 5A once for 300s. As soon as the current has been above 5A once, the genset is considered running properly, until it is stopped again. 

Alarm text:

[image]

The new setting is `com.victronenergy.settings/Settings/Generator1/Alarms/NoGeneratorAtDcIn` and will not be created for startstop instance 0 (`/Settings/Generator0/..`). The corresponding alarm path is `com.victronenergy.startstop1/Alarms/NoGeneratorAtDcIn`, which will only be valid when there is a DC genset connected. Gui-v2 uses the validity of this path to determine what setting to show.

mpvader, 2025-02-05:
nice feature! only one whish:

@philipptrenz / @ReinvdZee can you come up with a better name? "Detect generator at DC" is a bit strange wording, stranger than "at AC input".

philipptrenz, 2025-02-06:
Thanks, Matthijs!

@ReinvdZee How about "Alarm if DC generator is not delivering power"? It's about the same length as "Alarm when generator is not in autostart mode" and when activated, the toast provides the information, that it is only evaluated in the first 5 minutes after startup. So I think it should be clear enough.

ReinvdZee, 2025-02-06:
@philipptrenz 
> How about "Alarm if DC generator is not delivering power"? 

Sounds good. 

And the alarm text?
"DC generator is not delivering power"?

philipptrenz, 2025-02-06:
@ReinvdZee Sounds a bit strange, maybe we better use "providing" instead of "delivering" in both cases?

ReinvdZee, 2025-02-10:
@philipptrenz Ok, "providing" is fine with me.

@mpvader FYI

ReinvdZee, 2025-02-10:
Setting:

[image]

Alarm:

[image]

ReinvdZee, 2025-04-08:
Wording has been improved and merged. Closing

closed 2025-04-08 by ReinvdZee