victronenergy/venus #1419: Start/stop generator on losing CAN-bus BMS communication

ReinvdZee, 2025-01-31:
Generator start/stop already has startstop condition: "On loss of communication", which starts or stops the generator when the communication with the multi is lost, if a condition using the multi is enabled. Currently, this condition does not work with CAN-bus BMSes so the generator will not start when the communication with a CAN-bus BMS is lost. 

The "On loss of communication" condition works by having other enabled conditions return `None` when they fail. When this persists for x times, the communication required for that condition is considered lost. The battery conditions currently never return `None`, so the communication is not considered lost.

ReinvdZee, 2025-01-31:
Some details on how this works:

dbus_generator fetches the battery voltage, current and SOC values required for evaluating the conditions from the systemcalc paths `/Dc/Battery/..` Systemcalc fills these paths with the corresponding values of the currently selected battery monitor. A typical system with a vebus system and a managed BMS will thus have at least two battery monitors, where the managed BMS is selected when the selection is set to 'automatic'. The available battery monitors are listed in `com.victronenergy.system/AvailableBatteryServices` and the selected one is `/ActiveBatteryService`.

Not all battery monitors can be used as controlling BMS for DVCC. The ones that can be used are listed in `com.victronenergy.system/AvailableBmsServices`, and the selected one is `/ActiveBmsService`.

If a battery service disappears, systemcalc automatically picks the next best service for the job (at least, if the selection is set to automatic, which will be the majority of cases). That is also the reason the battery condition does not fail when the managed BMS disappears; the vebus battery monitor will just be used.

**On loss of communication for battery monitor**
Of course, we can consider the communication lost when the battery service changes, but that will also happen when the user manually selects a different battery monitor, or when the managed BMS shows up a little later than the vebus system. It would be difficult to determine what battery service is the one we want to monitor communication loss of. Best we can do is use the battery service selected when starting dbus-generator or when enabling the condition, but this leads to the "On loss of communication" condition to fire when the battery service changes from the vebus monitor to the managed BMS if the managed BMS shows up a little later. 

We can also look at `/com.victronenergy.system/ActiveBmsService` instead, which will become invalid when there is no DVCC-capable BMS anymore. However, this condition then won't work with DVCC-incapable battery monitors, like the vebus battery monitor and smartshunts. Besides, if there is a second DVCC-capable BMS in the system (e.g. a Lynx Smart BMS), that one will be selected instead and the path will still be valid.

**Explicit battery monitor selection**
There is a deprecated setting in Venus OS which used to allow the user to select the battery monitor to be used for the evaluation of the battery-related conditions. The path is `/Settings/Generatorx/BatteryService` and its default value is 'default'. This setting was removed from the UI later (only visible when the value is not 'default', so users on old systems can correct the setting).

If we reintroduce this setting, the user can explicitly set the battery monitor to look at for the battery conditions. Then, dbus-generator can also take action when this battery monitor disappears. If left to "Automatic" ('default' in dbus), the battery monitor as given by systemcalc will be used and the "On loss of communication" condition won't work for batteries.

philipptrenz, 2025-01-31:
@ReinvdZee I agree, that reviving/reusing `/Settings/Generatorx/BatteryService` to know which BMS communication to monitor is the best approach. This still allows to keep the global configuration at "Automatic" and, in case of failure, the system can automatically switch over to a secondary battery monitor while still alert about the failed communication.

ReinvdZee, 2025-02-05:
@philipptrenz 
> This still allows to keep the global configuration at "Automatic" and, in case of failure, the system can automatically switch over to a secondary battery monitor while still alert about the failed communication.

If set to 'Automatic', the system can indeed switch over to another battery monitor if the current one disappears. But I don't think we can conclude that the communication with the first battery monitor was lost. E.g. if the desired battery monitor shows up a little later while there already was one present, we trigger the condition because the communication with the first one was lost (while it was not, it was just not the preferred monitor). 

So I think we can only enable the "Communication lost" condition for battery monitors when the setting is explicitly set to a battery monitor. 

>**Explicit battery monitor selection**

@mpvader Agreed to reintroduce the explicit battery monitor selection for this?

ReinvdZee, 2025-02-12:
Discussed this with Matthijs: We're not going to do this because it's too complex and we don't want to reintroduce the explicit battery monitor selection.

closed 2025-02-12 by ReinvdZee