victronenergy/venus #1422: Phase rotation alarm not triggering in all inverter mode

Fllorent0D, 2025-02-12:
We’ve identified a inconsistency with the phase rotation alarm behavior. Currently, the alarm is only triggered when the inverter is set to `charger only` mode. However, when switching the inverter mode from `OFF` to `ON`, the phase rotation alarm does not activate, even if there’s an incorrect phase rotation sequence.

We expected, the phase rotation alarm to consistently trigger in all relevant inverter modes, including when the inverter is switched from `OFF` to `ON`, ensuring that phase rotation issues are promptly identified and resolved.

**Steps to Reproduce:**
- Set the inverter to charger only mode and observe that the `Alarms/PhaseRotation` activates correctly when a phase rotation issue is present
- Change the inverter mode from `OFF` to `ON` with the same incorrect phase rotation
- Notice that the `Alarms/PhaseRotation` does not activate, leaving the issue undetected

vejpasop, 2025-04-04:
On a Quattro or MultiPlus the phase rotation on `/Alarms/PhaseRotation` goes to invalid due to loss of VE.bus communication when the Quattro/Multi is switched `OFF`.  When switched to `ON` the item `/Alarms/PhaseRotation` will be set back to 1. This re-triggers the "Phase rotation" warning.

On a MultiPlus-II the behavior was a bit different. Since on these models as long as AC input is connected the MP-II keeps VE.Bus communication alive. So when switching the MP-II to `OFF`  the phase rotation warning remains active, and therefore does not get re-triggered  as long as AC-in is connected.

I changed this in mk2-dbus v3.80. 
Now when you switch to `OFF` or to `INVERTER ONLY` the phase rotation warning is cleared so it can be re-triggered when switching to `ON` or `CHARGER ONLY`.
This version will be listed for inclusion in Venus V3.60

closed 2025-04-18 by vejpasop

tomcauf, 2025-08-27:
Hello,

The phaseMapping alarm behavior is quite interesting. Following your changes, we tried to recreate a phase rotation alert, but we encountered some issues.

We tested it on a setup with:
- 3 MultiPlus-II 48/3000/35-32 inverters (ACOut2 switchable) ID:2629 running version `v556`
- Cerbo-S GX ID:C014 running version `v3.64`
- (On three-phase 400V)

For the test, I only tried with everything set to `ON`. As shown in the attached photo, I first disconnected from the grid at `8:32:17` to perform a phase rotation (swapping L1 and L2), then reconnected to the grid and received the phaseRotation alarm, which was great, but… it lasted only 16 seconds even though I hadn’t made any further changes. So, the system detected a phase rotation but quickly cleared the alert, even though the issue was still present.

After that, I restored the correct phase rotation and tried again later with other combinations like L1 - L3, L2 - L3, etc., with different waiting/correction times and repeated errors—many scenarios—but I never received the alarm again, as shown in the photo.

Could you provide more details about why it didn’t work and help find a possible solution to this problem to get the alarm?

[image]