victronenergy/venus #1423: VE.Bus: VRM installation overview on/off status can be incorrect when multiple VE.Bus systems are connected

vejpasop, 2025-02-14:
Discovered on Venus V3.53.

When Multiple VE.Bus systems are available the VRM installation overview can present the wrong on/off state if one of the systems is switched off.

[image]

[image]

[image]

izak, 2025-02-14:
I looked into this superficially already. The systemstate reflects (or should reflect) the state of the "main" inverter, the one plugged in on the built in port. If that works correctly, then there is no bug in systemcalc: The systemstate will reflect off, but there might still be some loads. Not ideal, just the way it is right now.

So first step is to try and reproduce it, and determine if this is in systemcalc, or in VRM.

izak, 2025-04-10:
I've confirmed that it does reflect the system state of the main inverter (plugged in on the built-in port). On Victron hardware (excluding Maxi-GX) it won't even allow using the state of the second inverter for the system state. Only the built-in mk2 is allowed.

On something like a Raspberry Pi, or a Maxi GX, it will reflect the state of the unit with the lowest DeviceInstance.

Result: could not reproduce. Also can't see an issue in the code.

@vejpasop, do you have any more details about this, or do we ask team VRM?