victronenergy/venus #1433: dbus api for injecting a notification into venus-platform

ReinvdZee, 2025-04-03:
Venus-platform watches the alarm/notification paths of different services and inserts a notification in the queue when a service issues one. To support scripts etc to show a notification on GUI-v2, we only need a means to inject one into the queue venus-platform maintains. 
The venus-platform dbus API needs to be extended with a means to inject a notification of any type directly into venus-platform.

ReinvdZee, 2025-04-03:
To inject a notification into the venus-platform queue, we need the following information:

- Notification text
- Device name
- Notification type

I'd suggest adding one new path to the API to support this:
`/com.victronenergy.platform/Notifications/Inject`

With the following syntax:

`"<type>:<name>:<description>"`

Where
- 'type' is an integer representing the notification type, 0 for warning, 1 for alarm and 2 for information
- 'name' is the name of the device, which will be visible on the toast and in the notification center.
- 'description' is the notification message itself.

Drawback of this method is that names with a colon are not supported because the colon is used as separator. But I don't think that's a problem. 

We can also go with a json like representation:

```json
{
    "type": 0,
    "name": "Some name",
    "description": "Some description"
}
```

But I think that unnecessarily complicates the serialization and parsing of the message.

another approach is to add three paths and an action path under `/Notifications/Inject/` but I don't think that's necessary.

@philipptrenz @mpvader any objections?

philipptrenz, 2025-04-03:
@ReinvdZee 

I think having a single dbus path to "inject" a notification is a good idea. And using JSON is the nicer solution, but not so easy to handle in shell scripts and such. So yes, the simpler syntax makes more sense to me.

Regarding the type, I'm not fully aware of the latest changes to the Notifications API, but maybe it would make sense to follow the `[ewi]` format for 3rd party services, like generators?

mr-manuel, 2025-04-03:
Keep in mind, that the description could also contain a colon `:`, but that should not be a problem if the whole string is quoted?

mpvader, 2025-04-03:
Simple syntax 👍; otherwise no comments

The node-red users will also be happy with this

dirkjanfaber, 2025-04-06:
If not using json, I would consider using a different character than the colon, e.g. the `|` or the `;` or even `\t`. 
Or, more ideally, allowing to escape the separator by a slash, so it can still be used.

ReinvdZee, 2025-04-07:
@mr-manuel 

> Keep in mind, that the description could also contain a colon :, but that should not be a problem if the whole string is quoted?

Since the description is at the end of the string to parse, it can contain a colon (or whatever we use as separation character). We just need to look at the first two colons.

ReinvdZee, 2025-04-07:
@philipptrenz 

> Regarding the type, I'm not fully aware of the latest changes to the Notifications API, but maybe it would make sense to follow the [ewi] format for 3rd party services, like generators?

I thought of using this as well but it does not fit straight away because it uses codes to denote the error/warning. The format is: `<man>:[ewi]-<code>`, which we would then need to change to `<name>:[ewi]-<description>`. I think the proposed syntax is a bit easier to use because it only uses one separator character.

philipptrenz, 2025-04-07:
> I think the proposed syntax is a bit easier to use because it only uses one separator character.

Fine with me!

ReinvdZee, 2025-04-07:
@dirkjanfaber, @philipptrenz, @mr-manuel, @mpvader 

> If not using json, I would consider using a different character than the colon, e.g. the | or the ; or even \t.
Or, more ideally, allowing to escape the separator by a slash, so it can still be used.

I think using JSON unnecessarily complicates this for users. I agree that we can pick a less common separation character to use for this.

If there are no further objections, I'll use the tab character `\t`:

`<type>\t<name>\t<description>`

I don't think escaping this character is necessary as I don't expect tabs to be used in the name or description.

ReinvdZee, 2025-04-07:
@mr-manuel, 

I think indeed that this is out of scope for this API. It will also be conflicting with the current behavior of venus-platform, because it will already engage the buzzer when the injected notification is of type 'alarm' and the the 'Audible alarm' setting is enabled.

So for 'system critical notifications', just push an alarm type notification and venus-platform takes care of the buzzer if enabled by the user.

philipptrenz, 2025-04-07:
I agree with Rein, that's also how we discussed it last week together with Matthijs: If needed, access the Buzzer directly.

ReinvdZee, 2025-04-07:
I'm working on the implementation of this and realized something: notifications are considered active as long as the source path is held in the active state. However, when a notification is injected with the mechanism above, it cannot be deactivated as there is no source path. After injecting, the `/Inject` path is cleared right away so a next notification can be injected.

An easy solution for this is to set the newly pushed notification to inactive straight away. The user still has to acknowledge it for it to disappear. It will look like this:

[image]

Only problem with this is with alarms; the buzzer and alarm relay are only engaged when the alarm source is active. So the relay and buzzer will not be engaged.

If we want to use this for notifications only it should be fine. 

Otherwise, we need to come up with a way to let the user set the notification to inactive again, or we don't clear the `/Inject` path and set the notification to active as long as the path contains a value. But that only allows for one notification at a time.

mr-manuel, 2025-04-22:
Currently an additional problem, when injecting notifications via this api, is that no popup shows up, but only the small icon near the clock. This makes the notification not that much visible to the user as it should.

## Injected notification

[image]

## Other notification

[image]

ReinvdZee, 2025-04-24:
> Currently an additional problem, when injecting notifications via this api, is that no popup shows up, but only the small icon near the clock. This makes the notification not that much visible to the user as it should.

As discussed this morning in the meeting; this should be improved. The notification is set to inactive right after injecting it because [of this](https://github.com/victronenergy/venus/issues/1433#issuecomment-2783211649). Still, the GUI should show the toast notification because the notification is unacknowledged.

I created a GUI-v2 issue for this: https://github.com/victronenergy/gui-v2/issues/2073

MikeTrahearn-Qinetic, 2025-05-16:
Would this also work across MQTT if relevant (given any appropriate adjustments to the path)?

MikeTrahearn-Qinetic, 2025-05-19:
For my own benefit; quick note on working Javascript parsing of above proposed \t delimited notification injection.
```
let text = "1\tdeviceName\tDescription\tincluding\tdelimiter";
let first = text.indexOf("\t");
let second = text.indexOf("\t", first + 1);

let type = text.slice(0, first)
let deviceName = text.slice(first + 1, second)
let description = text.slice(second + 1, text.length)
```

MikeTrahearn-Qinetic, 2025-05-28:
See updated comments in https://github.com/victronenergy/gui-v2/pull/2156 relating to the required modifications for Toasts to support injected notifications.