victronenergy/venus #1444: When running GUIv2, not connecting a display confusingly gives the GUI version choice in VRM

wiebeytec, 2025-04-28:
When a GX is configured to run GUIv2, but there is no display connected, it starts GUIv1 instead. The git commit message of `/usr/bin/start-gui` explains this as: "Starting gui-v2 only makes sense when there is a screen, since unlike gui-v1 it does not have VNC support."

This causes confusion: by not connecting a display, you get the GUI version prompt back on VRM. But users don't know why with some of their GXs they get this prompt on VRM, and with some they don't.

Not sure what the solution is. It may require some discussion, between choosing to change things in VRM, or never starting VNC when GUIv2 is selected in Venus, which in turn defeats the purpose of starting GUIv1 at all. Might as well start nothing then.

mpvader, 2025-08-19:
@wiebeytec shall we just close this? I don't hear or see any confusion recently.

wiebeytec, 2025-09-24:
@mpvader another confused user in support case 16379. They saw classic GUI on VRM as option on other installations, without a screen, so expected to see classic GUI on VRM as well for an installation _with_ a touch GX set to run GUIv2.