victronenergy/venus #1448: Use battery temperature reading for dbus-tempsensor-relay

ReinvdZee, 2025-04-30:
In venus OS, its possible to control a relay based on temperature readings. However, currently only dedicated temperature services show up in the list of temperature sensors to choose from:

[image]

dbus-tempsensor-relay should support the temperature reading provided by `com.victronenergy.battery` services as well so users can use the internal temperature measurement of the NG lithium batteries to switch a relay powering a heatpad attached to the battery.

martinbosma, 2025-05-01:
Related to https://github.com/victronenergy/venus/issues/1417

ReinvdZee, 2025-05-07:
I drafted an implementation for this. dbus-tempsensor-relay uses the fourth field of `com.victronenergy.temperature.x` (so fourth field is 'x'), as identifier for settings and to list the temperature sensors in the `com.victronenergy.temprelay` service. I extended this with allowing services of type `com.victronenergy.battery.x` as well, providing the path `/Dc/0/Temperature` is valid.

This also requires a small change to the GUI to let it find the service.
[image]

[image]

The `/Dc/0/Temperature` path is used for the temperature reading. In case of the Lynx Smart BMS NG, there is also the path `/System/MinCellTemperature` available, which could be useful when a heatpad must be switched based on the lowest temperature in the battery. I'm not sure how much the min, max and `/Dc/0/Temperature` (which is an average?) normally deviate, but I'd say its sufficient to use `/Dc/0/Temperature` for the temperature reading.

@mpvader Agreed to use `/Dc/0Temperature` for the readings? Or should we provide a drop down box to select the temperature reading out of min,max and the average?

ReinvdZee, 2025-05-12:
As just discussed: The minimum cell temperature path (`/System/MinCellTemperature`) is used for the temperature readings, if available. If not available, the `/Dc/0/Temperature` path is used. 

A battery will be listed as a temperature service whenever it has `/Dc/0/Temperature`.

closed 2025-05-13 by jhofstee