victronenergy/venus #1454: Venus OS on Nano Pi Neo

jemarr, 2025-05-09:
Hi.

I try to install Venus on Nano Pi Neo that is almost the same as Nano Pi Neo Core included in Multiplus II Gx card but without eMMc memory. I’ve downloaded this image “venus-install-sdcard-nanopi-20250403132010-v3.55.img” but there aren’t any response from my NanoPi. Connecting a console to the Nano Pi seems that system wants to initialize the eMMc at boot and stuck in several errors as there is any eMMc memory and finishes with ==> command prompt of u-boot waiting for commands.

Is Venus Os compatible with Nano Pi Neo (No CORE version) ???

Thanks

mpvader, 2025-05-12:
Hi @jemarr , no it is (apparently) not compatible; and not something we'd be interested in to invest time in, sorry.

closed 2025-05-12 by mpvader

jemarr, 2025-05-12:
Ok. Thanks.

danielprein, 2025-05-14:
How about the NanoPi Neo Air? That one has eMMc and WiFi. Would that one work instead of the NanoPi Core?