victronenergy/venus #1465: Missing dbg opkg debug symbols.

ieb, 2025-05-30:
Not sure if this is hte right place to ask, but here goes.....

Missing dbg packages from https://updates.victronenergy.com/feeds/venus/release/packages/

I have discovered a memory leak in the Victron Modbus RTU stack below the Python layer which I am trying to locate using https://github.com/rogerhu/gdb-heap, however the Victron Energy opkg repository says it has dbg symbols in its catalogue (present in opkg list), but they are not present.  Was that intentional or an oversight ?

Specific error is below. I have checked all the packages mentioned are in the catalog

```
root@nanopi:/data# opkg install libc6-dev libc6-dbg libglib-2.0-dbg python3-ply

Installing libc6-dbg (2.31+git0+2d4f26e5cf) on root

Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libc6-dbg_2.31+git0+2d4f26e5cf-r0_cortexa7hf-neon-vfpv4.ipk.

Collected errors:

* opkg_download_backend: Failed to download https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libc6-dbg_2.31+git0+2d4f26e5cf-r0_cortexa7hf-neon-vfpv4.ipk, wget returned 8.

* opkg_install_pkg: Failed to download libc6-dbg. Perhaps you need to run 'opkg update'?
```

For background, I have already verified that there is no leak in python objects or variables (using gc, _tracemalloc and gippy3) and I have a core dump of the process. Typically standard python services connecting to dbus have RSS around 30MB, but this one grows to 128MB after 24h, and if left wll cause the watchdog to reboot the OS. Since the only major difference is the ModbusRTU and serial connectors in C code, I suspect there is a leak there or its always possible I am doing something wrong.

I could restart every 24h, but that is a hack, and none of the other python services leak anything. I believe the leak is my problem to solve, but it would be easier with the debug symbols.  Based on telemetry going to Grafana I think it might be related to background scanning for new modbus RTU devices, or an error state failing to free some native buffers inside the Modbus RTU code.

If this is not the right place to ask.

mpvader, 2025-05-30:
Hey that fix is in that fix is included in v2.5.3 which we're using since Venus OS v3.60~70

ieb, 2025-05-30:
Looks like I have 3.54

```
/var/log/messages.2:May 30 02:28:46 nanopi cron.info CROND[9109]: (root) CMDOUT (installed: 20250128120635 v3.54)
/var/log/messages.2:May 30 02:28:46 nanopi cron.info CROND[9109]: (root) CMDOUT (available: 20250403132010 v3.55)
```

Does a Multiplus II GX normally get upgraded automatically ?

Also are the dbg packages available?

I am only guessing that 2.5.0 might fix the problem.

mpvader, 2025-05-31:
See manual for how to update Venus OS. v3.60~xxx are beta versions. Further details here: https://community.victronenergy.com/t/venus-os-v3-60-84-available-for-public-testing/30146

ieb, 2025-06-01:
2.5.3 did not eliminate the problem and as you say, switching to 3.x is non trivial as it requires Python 3.10 and the API changed.
If the dbg symbols are not available, I will have to live with a restart every 24h to avoid the watchdog going oom and restarting the OS.  Either that, or find/write an alternative modbus implementation.  Assuming that ModbusRTU is the problem and thats not confirmed.

mpvader, 2025-06-01:
Have you installed the latest v3.60 beta version and checked that for a memory leak?

I think that is the first thing todo before taking this further.

ieb, 2025-06-11:
FYI, I have located the leak by bysecting calls down to C code. The leak is in (or below) the C impl of select.select used in PySerial. There is nothing wrong with either PyModbus (not using async IO) or PySerial in the versions current release versions. All versions of the posix serial class in PySerial exibit the same behaviour, and a rewritten version using os.read on the file descriptor only also leaks.   On that basis it may not be worth doing the work to port code forward to the latest version of PyModbus in your 3.70 release.

I have looked at https://github.com/python/cpython/blob/main/Modules/selectmodule.c and are no commits back to 3.8 that look like they address this issue which indicates it may be a problem in the ch341 kernel driver which would be specifc to the hardware I have plugged in to the USB ports on the MultiPlus. I didnt go deeper than that.

The behaviour is about 1KB of memory is not freed every time a call to select.select or os.read where the file descriptor is a serial port. At a standard update rate, this results in the RSS growing by 10MB/h, however only while a modbus unit is non-responsive. My solution to minimise the impact was to detect this and backoff the rate so that the leak rate drops from 10MB/h to more like 100KB/h if timeouts are happening. I also now monitor RSS and exit when it reaches 32MB letting Supervise start a fresh process.