victronenergy/venus #1473: v3.60 new modifications check doesn't work on a Venus GX?

mpvader, 2025-06-10:
Reported here: https://community.victronenergy.com/t/venus-os-v3-60-released-officially/30146/179?u=mpvader

System with a Venus GX: https://vrm.victronenergy.com/installation/28724/overview

mpvader, 2025-06-10:
if not trivial to solve, then we have to see how to handle this. Venus GX are EOL.

mr-manuel, 2025-06-11:
As soon as I have the Venus GX device I will investigate.

mr-manuel, 2025-06-14:
Tested now also on Color Control GX.

```
root@ccgx:~# fsmodified /
unknown
```

and

```
[    0.000000] Kernel command line:  root=ubi0:rootfs1 rootfstype=ubifs ubi.mtd=12 omapdss.def_disp=lcd omapfb.vram=0:2M@0x8fe00000 fbskip fbcon=vc:63 mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot),256k(env1),256k(env2),1m(u-boot2),256k(bootparms),768k(splash),6m(kernel1),6m(kernel2),496m(ubisystem) 5
[    0.000000] Unknown kernel command line parameters "5 mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot),256k(env1),256k(env2),1m(u-boot2),256k(bootparms),768k(splash),6m(kernel1),6m(kernel2),496m(ubisystem)", will be passed to user space.
```

`dmesg`

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.12.23-venus-5 (oe-user@oe-host) (arm-ve-linux-gnueabi-gcc (GCC) 13.3.0, GNU ld (GNU Binutils) 2.42.0.20240723) #1 Fri Jun 13 16:34:54 UTC 2025
[    0.000000] CPU: ARMv7 Processor [411fc087] revision 7 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Color Control GX
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 16 MiB at 0x8ec00000 on node -1
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000008fdfffff]
[    0.000000]   HighMem  [mem 0x000000008fe00000-0x000000008fefffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x000000008fdfffff]
[    0.000000]   node   0: [mem 0x000000008fe00000-0x000000008fefffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000008fefffff]
[    0.000000] OF: reserved mem: 0x8fe00000..0x8fffffff (2048 KiB) nomap non-reusable framebuffer@8fe00000
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM3517 ES1.1 (l2cache sgx neon)
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Kernel command line:  root=ubi0:rootfs1 rootfstype=ubifs ubi.mtd=12 omapdss.def_disp=lcd omapfb.vram=0:2M@0x8fe00000 fbskip fbcon=vc:63 mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot),256k(env1),256k(env2),1m(u-boot2),256k(bootparms),768k(splash),6m(kernel1),6m(kernel2),496m(ubisystem) 5
[    0.000000] Unknown kernel command line parameters "5 mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot),256k(env1),256k(env2),1m(u-boot2),256k(bootparms),768k(splash),6m(kernel1),6m(kernel2),496m(ubisystem)", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65280
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] IRQ: Found an INTC at 0x(ptrval) (revision 4.0) with 96 interrupts
[    0.000000] Clocking rate (Crystal/Core/MPU): 26.0/332/600 MHz
[    0.000000] OMAP clocksource: 32k_counter at 32768 Hz
[    0.000000] clocksource: 32k_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 58327039986419 ns
[    0.000000] sched_clock: 32 bits at 33kHz, resolution 30517ns, wraps every 65535999984741ns
[    0.002807] TI gptimer clockevent: always-on 13000000 Hz at /ocp@68000000/target-module@48318000
[    0.004547] Console: colour dummy device 80x30
[    0.004638] printk: legacy console [tty0] enabled
[    0.008941] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.009094] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.009216] ... MAX_LOCK_DEPTH:          48
[    0.009307] ... MAX_LOCKDEP_KEYS:        8192
[    0.009399] ... CLASSHASH_SIZE:          4096
[    0.009490] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.009613] ... MAX_LOCKDEP_CHAINS:      65536
[    0.009704] ... CHAINHASH_SIZE:          32768
[    0.009796]  memory used by lock dependency info: 1925 kB
[    0.009918]  per task-struct memory footprint: 1536 bytes
[    0.010162] Calibrating delay loop... 597.60 BogoMIPS (lpj=2988032)
[    0.123138] CPU: Testing write buffer coherency: ok
[    0.123443] CPU0: Spectre v2: using BPIALL workaround
[    0.123565] pid_max: default: 32768 minimum: 301
[    0.124084] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.124298] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.153259] Setting up static identity map for 0x80100000 - 0x80100060
[... 259 lines cut ...]
[   56.753631]  join_cmd_hdl [8192eu] from rtw_cmd_thread+0x31c/0x364 [8192eu]
[   56.755340]  rtw_cmd_thread [8192eu] from kthread+0xc4/0xe0
[   56.756317]  kthread from ret_from_fork+0x14/0x28
[   56.756408] Exception stack(0xd070dfb0 to 0xd070dff8)
[   56.756469] dfa0:                                     00000000 00000000 00000000 00000000
[   56.756530] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   56.756561] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   56.756927] ---[ end trace 0000000000000000 ]---
[   80.649963] ubi0: fixable bit-flip detected at PEB 2144
[   80.650085] ubi0: schedule PEB 2144 for scrubbing
[... cut ...]
[   84.797760] ubi0: scrubbed PEB 1675 (LEB 0:593), data moved to PEB 2144
[   99.051361] can: controller area network core
[   99.051818] NET: Registered PF_CAN protocol family
[  100.322204] ti_hecc 5c050000.can can0: setting CANBTC=0x30049
[  103.967468] can: raw protocol
[  124.309722] ubi0: fixable bit-flip detected at PEB 2211
[  124.309844] ubi0: schedule PEB 2211 for scrubbing
[  124.318298] ubi0: fixable bit-flip detected at PEB 2211
[  124.450988] ubi0: scrubbed PEB 2211 (LEB 0:1100), data moved to PEB 2094
```