victronenergy/venus #1483: Feature Request: Support for Hybrid Inverter and/or AC-Coupled Battery Storage

pellenbeck, 2025-06-16:
### Context
A third-party hybrid inverter is currently implemented in Venus OS solely as a PV inverter (by using [venus-os_dbus-mqtt-pv](https://github.com/mr-manuel/venus-os_dbus-mqtt-pv)), causing nighttime battery discharge and cross-inverter charging to be misclassified as PV production by night or grid import in the event of existing PV surplus. This distorts VRM Portal data and leads to inaccurate PV and consumption forecasts.

### Proposed Requirement
Add an explicit device type for “Hybrid Inverter” and/or a separate “AC-Coupled Battery Storage” in Venus OS, enabling distinct data streams for PV generation and battery operation.

### In Case of Hybrid Inverter
Add the option to save data in **com.victronenergy.pvinverter** for a connected battery with at least measured value for the current power.

### In Case of AC-Coupled Battery Storage
Create a new device type AC-Battery (example naming).
Add the option to save data in **com.victronenergy.acbattery** for a connected battery storage with at least measured values for the current power (L1, L2, L3 / Sum). 

Please review and integrate these enhancements to ensure precise VRM Portal reporting and forecasting.

spacecabbie, 2025-06-22:
The repository you referenced is not built or supported by Victron. Knowing Mr. Manuel's work, it is most likely the best implementation you will find for your inverter.  Also, the chance that Victron will support this is slim.

The reason is simply that these inverters, even though they superficially appear to run the same protocols, unfortunately have vastly different per-inverter implementations. There seems to be some progress toward standardization, but they are far from it. Even if Victron wanted to, it is impossible to support something that changes almost per unit, knowing how busy the developers are. I just wanted to provide this feedback.

That being said, I would love to see an official integration, obviously.

mpvader, 2025-06-22:
No interest; closing.