victronenergy/venus #1487: VRM Instance ID on VE.Bus devices changes on every reboot of VenusOS

gyrex, 2025-06-21:
I've got 3x Multiplus-ii 5kVA inverters configured in 3-phase connected to a Raspberry Pi 3B+ running VenusOS v3.60.

There's a well known issue where the VRM instance ID changes on every reboot of the Raspberry Pi and oscillates between 288 & 289. The issue is that VRM historical data is split between these 2 IDs and any custom widgets that I've created for VE.Bus datasets needs to be recreated whenever the RPi is rebooted due to the VRM instance ID changing. 

Is there any way to fix this issue or hard code a VRM instance ID to VE.Bus devices to avoid this?

Forums threads where this has been discussed:

https://community.victronenergy.com/t/explaining-changing-vrm-instance-numbers/16935
https://communityarchive.victronenergy.com/questions/235685/raspberry-pi-4b-venus-os-the-multiplus-481600-show.html

mpvader, 2025-08-19:
Hi, yes, there is an issue for this elsewhere. But its not super trivial and there is lots of work around vebus all the time, so limited capacity.

So, on the list, hopefully gets done some day but can't make any promises here - sorry.

closed 2025-08-19 by mpvader

chaneych70, 2026-10-12:
Any news with this issue ?

bontas69, 2026-10-12:
Try this:
Open VRM portal, go to devices and delete all multiplus instances.
Then reboot venus OS and maybe multiplus get assigned with the old instance number.
In my case the 288 is the basic/main/first instance. Had also 289 and 290. After reboot seems to get 288 again.

chaneych70, 2026-10-12:
Thanks, I managed to use this trick but that's not a real solution ! 😉