victronenergy/venus #1488: Feature Request: Add vcgencmd to venus os to allow monitoring RPi hardware.

spacecabbie, 2025-06-22:
Hi,

I know we can monitor base sensors, temperature, load, etc. But with vcgencmd, we can get throttling information; it also communicates directly with the CPU, giving accurate, up-to-date information.

The benefit of having it in the core OS allows for the development of integration of these values directly into the GUI on a later occasion.

If this is not desirable or cannot or will not be implanted. Please close the issue.

mpvader, 2025-06-22:
hey, sorry but no - we're not going to do this.

But you can build it yourself if you want to. I'd advise to build/compile it on your rpi, rather than cross compile it.

here is a list of commands I did:

```
/opt/victronenergy/swupdate-scripts/resize2fs.sh 
opkg update
opkg install packagegroup-core-buildessential git
git clone --depth 1 https://github.com/raspberrypi/utils.git
cd ./utils/cgencmd/
gcc ./vcgencmd.c
chmod a+x ./a.out
```

and now a.out is your executable. doesn't work for me since I'm on a Cerbo:

```
root@einstein:~/utils/vcgencmd# ./a.out 
Can't open device file: /dev/vcio
Try creating a device file with: sudo mknod /dev/vcio c 100 0
```

closed 2025-06-22 by mpvader

spacecabbie, 2025-06-24:
> Hi, good that it helped. What exactly do you mean? Which issues? Probably / possibly yes :-)

A bit about my background:
I am a retired Dutch customer service manager/system operator now living in Portugal. I am atypical, with one foot in developer thinking and one in customer thinking. I do basic-level programming—not skilled enough to develop solutions from scratch but capable of understanding concepts and performing basic troubleshooting. This has given me a unique perspective, bridging both worlds. Nowadays, they might call it a GUI/interface developer role, but for me, it’s always been about aligning customer viewpoints with developer perspectives. I’m sure you’ve heard arguments like, “You can view information X in this menu; click here and there.” Logical for a technical mind, but not for a consumer. 

Things like that.What I’m getting at is simple: What do you need? You don’t know me, so we’d need to build trust, and I don’t know how you work. For me to dive in and start “cleaning” issues would be disastrous at this stage. Instead, my philosophy is: How can I help you keep issues clean and manageable? I may not contribute to open-source coding, but I have skills that let developers focus on coding rather than dealing with unimportant tasks.I’ve noticed many open issues in the repository. Given Victron’s popularity, I suspect quite a few are essentially support questions rather than true issues. 

My suggestions (just suggestions):
First, implement templates for issue creation (e.g., Bug/Feature Requests) with required sections to help you quickly assess requests. A great example is: https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues. Open an issue there to see it in action.
Templates eliminate frustrating tickets with insufficient information, which often get left open and forgotten. 

A few examples:
https://github.com/victronenergy/venus/issues/1460 – Likely a user issue, not a bug, possibly on an older version (v3.52).
https://github.com/victronenergy/venus/issues/1479 – Not something developers should handle.
https://github.com/victronenergy/venus/issues/1461 – Feature request example.

Next, evaluate all current feature requests, close them or adopt them to the new template. Or, I’m personally a fan of this: set up a feature tracker like DirectAdmin uses: https://feedback.directadmin.com/.
This prevents duplicated features and shows which features are important to the community.

I would also suggest a timer of sorts: when a ticket has been open for XX amount of time, the user gets an email notifying them the ticket will be closed in one day, followed by the closing of the ticket and a final email.

I want to stress one more time: these are suggestions. In the end, I want to help out, and I know how expensive developers are. It’s really my way of contributing to open source. Anyway, I am not a great writer. I’m better at communicating in person. I hope I’ve been able to explain how I could and want to help.

My email: me(a)hhaufe.eu 
if you want to further discuss this so we don’t pollute this ticket 😄