victronenergy/venus #1490: Configuration gets lost for BLE (tank) sensor

dirkjanfaber, 2025-06-23:
Changes in `/Capacity`, `/RawValueFull` and `/RawValueEmpty` for a BLE tank sensor get lost. It looks like the changes don't reach `Settings/Devices/mopeka_dc4ce45198f4/Capacity` in `com.victronenergy.settings`, while changes in `FluidType` and `CustomName` do get there. Looks like float values don't get stored.

This is reproducible on 3.60 and possibly introduced in version 0.23 of [dbus-ble-sensors](https://github.com/victronenergy/dbus-ble-sensors).

mpvader, 2025-06-25:
Isn’t it the other way around?

You configured those setting using v3.60; which would seem like it worked, but actually weren’t properly stored.

Now you updated, rebooted, settings were lost (bug of v3.60).

Please try making the config, reboot, and see if your settings stick over the reboot

mpvader, 2025-06-25:
👍 thank you for the quick confirmation!