victronenergy/venus #1494: Look into not starting gui-v1 on Cerbo without display

mpvader, 2025-06-25:
Right now gui-v1 still starts, even when user has selected gui-v2 in the menu, as per screenshot below.

Benefit is to save cpu. Its not so bad as rending the moving ants all the time, display goes to sleep. But still its 8 to 11% unnecessary CPU usage as per what I've seen on system.

[image]

[image]

mpvader, 2025-06-25:
https://github.com/victronenergy/meta-victronenergy-private/blob/master/recipes-ve/start-gui/start-gui

jhofstee, 2026-08-27:
It is not a bug, it is done on purpose. It allows gui-v1 and gui-v2 to be used a the same time and allows a choice in VRM which UI to use. It would be interesting to know if users are using that or use "Remember my choice for this installation" most of the time. For the LAN case it is a good idea to introduce it early in a release cycle or make it depend on a setting (VncLocal e.g.). Anyway, moving it to v3.90 now, but it is a good idea to free up that cpu usage when not used at all.