victronenergy/venus #1508: Add support for SeeLevel II Bluetooth integration for 709-BTP3

cgoudie, 2025-07-24:
The SeeLevel 709-BT is a SeeLevel device which broadcasts tank information over bluetooth. This feature request is to integrate it into the Victron Cerbo GX MK2.

In my Airstream, there is no easy way to get a cable from the wall the Seelevel is in to the Cerbo. It would involve removing most of the kitchen.

Decoding the packets received on bluetooth looks fairly straightforward: https://github.com/custom-components/ble_monitor/issues/1181 

---

Garnet 709BT BLE Technical Document
The Garnet 709BT hardware supports Bluetooth Low Energy (BLE), and is configured as a Broadcaster transmitting advertisement packets. It continuously cycles through its connected sensors sending out sensor data. No BLE connection is required to read the data, instead the smartphone application implements Receiver functionality to receive these packets.

When the application receives these packets it must decode the Manufacturer Specific Data to determine the Garnet Sensor Data. The first 2 bytes represent a company identifier code.

Company ID 305 Cypress Semiconductor

The following bytes represent the Garnet Sensor Data payload:
Byte 0, 1, 2 Coach ID Unique number - hardware assigned
Byte 3 Sensor Number 0-255
0 = Fresh
1 = Black
2 = Gray
3 = LPG
4 = LPG 2
5 = Galley
6 = Galley 2
7 = Temp
8 = Temp 2
9 = Temp 3
10 = Temp 4
11 = Chemical
12 = Chemical 2
13 = Battery x 10
Byte 4, 5, 6 Sensor Data ascii characters representing data value or sensor status
Byte 7, 8, 9 Sensor Volume
Byte 10, 11, 12 Sensor Total
Byte 13 Sensor Alarm

---

The link above has code which can be used to read the signals, but the spec makes it plain how to read them.

Please integrate this so I can display the tanks on my Cerbo GX without having to spend thousands and thousands of dollars trying to run a wire.

This ticket looks a lot like #858 but for the SeeLevel II, and for tank sensors instead of temperature and atmosphere sensors.

cgoudie, 2025-08-31:
@mpvader Anything I can do to help?

cgoudie, 2025-10-04:
Helping things along, here are some samples from btmon. I know, my tanks didn't actually change levels so the data is all the same.
```
 HCI Event: LE Meta Event (0x3e) plen 43                       #1273 11.248701
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scannable undirected - ADV_SCAN_IND (0x02)
        Address type: Public (0x00)
        Address: 00:A0:50:8D:95:69 (OUI 00-A0-50)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): 709-BT
        Company: Cypress Semiconductor (305)
          Data: 8d9569034f504e30303030303030
        RSSI: -62 dBm (0xc2)

 HCI Event: LE Meta Event (0x3e) plen 43                       #1971 17.100710
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scannable undirected - ADV_SCAN_IND (0x02)
        Address type: Public (0x00)
        Address: 00:A0:50:8D:95:69 (OUI 00-A0-50)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): 709-BT
        Company: Cypress Semiconductor (305)
          Data: 8d9569034f504e30303030303030
        RSSI: -76 dBm (0xb4)

HCI Event: LE Meta Event (0x3e) plen 43                       #1987 17.227741
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scannable undirected - ADV_SCAN_IND (0x02)
        Address type: Public (0x00)
        Address: 00:A0:50:8D:95:69 (OUI 00-A0-50)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): 709-BT
        Company: Cypress Semiconductor (305)
          Data: 8d95690d31333730303030303030
        RSSI: -60 dBm (0xc4)
```

cgoudie, 2025-10-04:
8d95690020323030303030303030

8d 95 69  - Coach ID: 0x69958d
00        - Sensor: 0 (Fresh Water)
20 32 30  - Data: " 20" → 20%  ✓ NEW!
30 30 30  - Volume: "000" → 0 gallons (still zero)
30 30 30  - Total: "000" → 0 gallons (still zero)
30        - Alarm: '0'

On all the models I know of you can't configure the volume at all, so the volume and total are zero

cgoudie, 2025-10-04:
@mpvader @mansr 
I've sent you a PR with what I think is the correct code at: https://github.com/victronenergy/dbus-ble-sensors/pull/11

I've never done this before on my Cerbo. I would have built it and tried it myself except for https://github.com/victronenergy/dbus-ble-sensors/issues/10

I'm happy to take a built one and try it, or if you can grant me access I can try to build it myself.

mpvader, 2025-10-05:
Hi, nice! Can you send me an email? (Assuming you can find my email address somewhere)

cgoudie, 2025-10-08:
> Hi, nice! Can you send me an email? (Assuming you can find my email address somewhere)

Should be in your inbox ;)