victronenergy/venus #1515: raspberry pi 4 .... is wifi useless?

ortegafernando, 2025-08-08:
Hi, may be i have not read it but I havent succes to connect a venus os 3.64 in raspberry pi 4 to any wifi network. 

Is it possible? Or we can only use cable connection.

I have try 2.4ghz, 5ghz, and different type of passowrd. Not luck.

Could anybody confirm me that it is possible to use wifi?

Thanks.

kwindrem, 2025-08-08:
I normally run with a wired connection to my RPI test system but just ran a test and a WiFi connection to my house router worked fine just now running v3.70~25.and v3.64.

My router console says the RPI connected via 5 GHz.

ortegafernando, 2025-08-11:
Hi, if anyone has same problem. I think I have solved with a manual ip configuration and psk2 password type. I dont know if it is Pi4 fault or my openwrt router configurations.

Thanks a lot @kwindrem .