victronenergy/venus #1521: Use `/dev/serial/by-id/` (and endpoint serial numbers) instead of `/dev/ttyUSBx` for stable device mapping in Victron services

RafaelKa, 2025-09-01:
### Problem description

On Venus OS, USB device handling is unreliable when GX devices are connected via USB hubs or when running on SBCs such as Raspberry Pi.  
Observed issues:  

- **Device persistence**: e.g. a MultiPlus-II connected via MK3 does not appear consistently in VRM. Each reboot creates a new entry and leaves the old one as “not connected.”  
- **D-Bus mappings**: service names change on every reboot, making them hard to query or use reliably in scripts.  
- **Scalability**: adding more devices (MPPTs, SmartShunt, MultiPlus-II, custom BMS, etc.) increases instability and forces workarounds.  

Most probably these issues are less visible in Victron’s test environments (likely using stable USB controllers without hubs), therefore all those issues become low priority and/or are rejected by Victron?
But in real-world PV installations, USB-hubs and mixed hardware are common. 
Ironically, many third-party devices connected via USB–serial adapters behave more reliably because the adapters are excluded from `serial-starter` and already mapped via `/dev/serial/by-id/`.

---

### Impact

In a typical small installation with USB-Hub, the following issues occur:
- 2× Smart MPPTs → affected
- 1× SmartShunt → affected
- 1× MultiPlus-II → affected
- 1× BMS via USB-RS485 on [venus-os_dbus-serialbattery](https://github.com/mr-manuel/venus-os_dbus-serialbattery) → partially affected
   The [EXTERNAL_SENSOR_DBUS_DEVICE setting](https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/v2.0.20250729/dbus-serialbattery/config.default.ini#L210) not usable, because D-Bus mapping unreliable
- 1× RS485 adapter for Modbus → not affected (since excluded from `serial-starter` and mapped via `/dev/serial/by-id/`)

This results in frustration, unstable systems, and risk of misconfigured or unreachable devices.

---

### Proposed solution

- Use **stable symlinks** (`/dev/serial/by-id/`) instead of `/dev/ttyUSBx` for device detection and mapping.
- Derive **D-Bus service names from endpoint Victron device serial numbers** instead of USB port order.
  - Example: `com.victronenergy.*.HQ25123ABC` instead of `com.victronenergy.*.ttyUSBx`.
- Introduce a new path such as `/dev/serial/by-victron-serial-number/`,
  which would symlink to the corresponding `/dev/serial/by-id/` device.
  ```bash
  HQ25123ABC -> /dev/serial/by-id/usb-VictronEnergy_MK3-USB_Interface_HQ25345QHH6-if00-port0
  HQ22129XYZ -> /dev/serial/by-id/usb-Silicon_Labs_CP2108_Quad_USB_to_UART_Bridge_0001-if01-port0
  ```
- This guarantees stable, predictable mappings even after reboots or wiring changes.

### Rollout suggestion

- Provide as a **configurable option** in the current major release (opt-in).
- Make it the **default behavior** in the next major release.
- Ensure **backward compatibility** (non-breaking migration path).

---

### Summary:
 
Persistent USB mapping is a **reliability requirement**, not a nice-to-have.
Without it, even Victron’s own adapters behave unpredictably in real-world setups, 
while with it, systems remain stable, automation reliable, and VRM data consistent.

kwindrem, 2025-09-01:
A downside to using a serial number as the device ID is that of a device is replaced because the original fails, then previous relationships will be broken. At least, in this situation the new device is likely to be an orphan and require a configuration step. This is better than automatically assigning functionality to a device based on which USB port it is connected to and which order USB devices are discovered!

Also, non-Victron devices may not have a serial number.

nmbath, 2026-09-09:
## Where `/DeviceInstance` does and doesn't help

`/DeviceInstance` already works well for higher-level consumers that discover devices dynamically and don't care about the underlying D-Bus service name/path or which physical port something is on.

But even with `/DeviceInstance`, GPS can end up with multiple different `gps:N` instance settings over time, because it can move tty on reboot and there's no unique serial number or data point to key a single device instance to.

Anything that either interacts with the hardware directly or references `com.victronenergy.<service>.ttyUSBx` rather than resolving by instance still has problems:

- **Node-RED isn't fully exempt either** — `dbus-listener.js`'s `searchHaystack()` resolves by `deviceInstance` when it can, but falls straight back to the raw tty-based name (`return fallback`) if no live instance match is found, which is exactly the case when the instance itself has moved.
- **SignalK** pulls the D-Bus service name into its data model as the source device, so it can end up with multiple source references as the ttys move around.

Therefore even with `/DeviceInstance` we have multiple issues from not having a deterministic `ttyUSBx` value.

## The fix

Make the tty enumeration itself deterministic without any higher-level service needing to change. Within `serial-starter`, hash the serial+interface if it exists (consistent across both reboot and port moves), otherwise the udev path based key (consistent through reboots only) is hashed, to create `/dev/ttyUSB<HASH>`, designed to be collision-resistant — the hash width can be widened further if that's ever not enough in practice.

There is one limitation: for a device with no reported serial, moving it to a different physical port will change its hash — that's inherent to using physical position as the key when there's nothing else to anchor identity to.

## Tested

Live-verified on two GX units across gps-dbus, vedirect-dbus, and dbus-imt-si-rs485tc.

RafaelKa, 2026-09-13:
### Proof of concept for Deterministic & Idempotent `ttyUSBx` mappings for ttyUSBx devices.

After deep diving into the Venus OS kernel hardware initialization layer, 
I have designed and fully verified a **100% stable, fully dynamic, and update-resilient workaround**.

Instead of refactoring the `serial-starter` architecture to support `/dev/serial/by-id/`, 
this solution targets the asynchronous nature of the Linux kernel during cold boot (the "boot lottery") and enforces 
**strict chronological sequencing** of the physical USB ports starting at `ttyUSB0`.

---

### The Core Logic

Linux natively maps `ttyUSBx` nodes asynchronously based on which device responds first during early boot. 
To achieve absolute idempotency, we can use [Linux **USB Device Authorization Layer** (`authorized_default=0`)](https://docs.kernel.org/usb/authorization.html).

1. **Global USB Lockdown / Disabling all USB Ports:** 
   We tell the kernel at boot to discover physical USB hardware, 
   but **refuse to bind any drivers or assign TTY numbers**.
2. **Sequential enabling loop from main USB-hub[s] and then the physical USB-Ports 1 – n them one by one:** 
   A lightweight script running during early system startup (`rc.local` or at any other later point, that does not matter)
3. **Hotplug Reactivation:** Once the critical boot-sequence is locked into place (`ttyUSB0` to `ttyUSB<n>`),  
   the script must restore the native USB hotplugging capabilities for any devices plugged in during live operation.

---

### Step-by-Step Implementation

##### Note:

**That is a POC only!**, which works on my hardware(CM4 @WS-18606), which has 1-1 only.
So most probably the final or production ready implementation must do a little bit more(e.g. resolve all root USB-hubs dynamically and them dive in each USB-tree recursively).
I could not test the behavior on real USB-Hubs, because I do not have any and do not want any...

#### Step 1: Activate the Global USB Lockdown

Append the kernel parameter to your boot configuration file to prevent the kernel from automatically assigning TTY nodes to devices during early boot.

Open `/u-boot/cmdline.txt` and append ` usbcore.authorized_default=0` to the end of the **existing single line**:

```shell
dwc_otg.lpm_enable=0 console=serial0,115200 usbcore.authorized_default=0
```

#### Step 2: Create the sequential USB-Port authorization/re-enabling script

Create a new executable script at `/data/rc.local.d/idempotent_ttyUSBx_numbers_issue_1521.sh` (**let it be called by** `/data/rc.local` and make it executable:* `chmod +x /data/rc.local.d/idempotent_ttyUSBx_numbers_issue_1521.sh`
).

This script relies on zero hardcoded device IDs, paths, or serial numbers. 
It dynamically scans the physical bus topology (`1-1.*`), 
sorts it naturally, 
introduces a safe 1-second driver settling-delay per physical port, 
and reactivates Hotplugging at the very end:

```bash
#!/bin/bash
exec 2>&1
echo "=== \[IDEMPOTENCY START of ttyUSBx by real USB-Port number\] See: Venus Issue https://github\.com/victronenergy/venus/issues/1521 ==="

# Step 1: Wake up the main onboard USB hub
echo 1 > /sys/bus/usb/devices/1-1/authorized
sleep 1

ACTIVATED_PORTS=""

# Step 2 : Wake up the occupied USB-Ports sequentially
while true; do
    NEW_PORT_FOUND=0
    
[... 46 lines cut ...]
```log
root@raspberrypi4:~# ls -la /dev/ttyUSB*
crw-rw----    1 root     dialout   188,   0 Sep 13 23:32 /dev/ttyUSB0
crw-rw----    1 root     dialout   188,   1 Sep 13 23:32 /dev/ttyUSB1
crw-rw----    1 root     dialout   188,   2 Sep 13 21:48 /dev/ttyUSB2
crw-rw----    1 root     dialout   188,   3 Sep 13 23:32 /dev/ttyUSB3
crw-rw----    1 root     dialout   188,   4 Sep 13 23:32 /dev/ttyUSB4
crw-rw----    1 root     dialout   188,   5 Sep 13 23:32 /dev/ttyUSB5
crw-rw----    1 root     dialout   188,   6 Sep 13 23:32 /dev/ttyUSB6
crw-rw----    1 root     dialout   188,   7 Sep 13 23:32 /dev/ttyUSB7
crw-rw----    1 root     dialout   188,   8 Sep 13 23:32 /dev/ttyUSB8
crw-rw----    1 root     dialout   188,   9 Sep 13 23:32 /dev/ttyUSB9
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadTTL_00 -> ttyUSB2
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadTTL_01 -> ttyUSB3
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadTTL_02 -> ttyUSB4
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadTTL_03 -> ttyUSB5
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadVeDirect_00 -> ttyUSB6
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadVeDirect_01 -> ttyUSB7
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadVeDirect_02 -> ttyUSB8
lrwxrwxrwx    1 root     root             7 Sep 13 16:41 /dev/ttyUSB_TO_QuadVeDirect_03 -> ttyUSB9
```

Since all TTY mappings are mathematically bound to the physical layout of the board, 
VRM Device Instances never shift, 
ghost orphans are a thing of the past, and third-party tools (like `venus-os_dbus-serialbattery`) run with 100% uptime and deterministic device paths.