victronenergy/venus #1522: dbus-modbus-client frequent crashes and restarts

dirkjanfaber, 2025-09-02:
Noticed on several system s(3.6x and 3.7x) that the dbus-modbus-client logging shows several Exceptions and connections failures.

```
2025-09-02 05:47:31.348109500 pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.135:502)]
2025-09-02 05:47:31.872520500 INFO     [tcp:192.168.0.135:502:1] Device failed: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.135:502)]
2025-09-02 05:47:47.758950500 INFO     [tcp:192.168.0.135:502:1] Found EV charger: Victron Energy AC22NS
2025-09-02 05:47:48.035106500 INFO     registered ourselves on D-Bus as com.victronenergy.evcharger.evc_HQ2326WG34D
2025-09-02 05:57:07.838164500 INFO     [tcp:192.168.0.135:502:1] Device failed: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.135:502)]
2025-09-02 05:57:15.704498500 INFO     [tcp:192.168.0.135:502:1] Found EV charger: Victron Energy AC22NS
2025-09-02 05:57:15.938797500 INFO     registered ourselves on D-Bus as com.victronenergy.evcharger.evc_HQ2326WG34D
2025-09-02 05:59:30.276540500 Traceback (most recent call last):
2025-09-02 05:59:30.285348500   File "/opt/victronenergy/dbus-modbus-client/device.py", line 271, in dbus_write_register
2025-09-02 05:59:30.285392500     self.write_register(reg, val)
2025-09-02 05:59:30.286392500   File "/opt/victronenergy/dbus-modbus-client/device.py", line 132, in write_register
2025-09-02 05:59:30.286400500     self.write_modbus(reg.base, reg.encode())
2025-09-02 05:59:30.286773500   File "/opt/victronenergy/dbus-modbus-client/device.py", line 126, in write_modbus
2025-09-02 05:59:30.286780500     self.modbus.write_register(base, val[0], unit=self.unit)
2025-09-02 05:59:30.287113500   File "/usr/lib/python3.12/site-packages/pymodbus/client/common.py", line 92, in write_register
2025-09-02 05:59:30.287121500     return self.execute(request)
2025-09-02 05:59:30.287123500            ^^^^^^^^^^^^^^^^^^^^^
2025-09-02 05:59:30.287464500   File "/opt/victronenergy/dbus-modbus-client/client.py", line 32, in execute
2025-09-02 05:59:30.287470500     return super().execute(*args)
2025-09-02 05:59:30.287472500            ^^^^^^^^^^^^^^^^^^^^^^
2025-09-02 05:59:30.287816500   File "/usr/lib/python3.12/site-packages/pymodbus/client/sync.py", line 108, in execute
2025-09-02 05:59:30.287822500     raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
2025-09-02 05:59:30.288286500 pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.135:502)]
2025-09-02 05:59:30.808706500 INFO     [tcp:192.168.0.135:502:1] Device failed: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.135:502)]
2025-09-02 05:59:39.368612500 INFO     [tcp:192.168.0.135:502:1] Found EV charger: Victron Energy AC22NS
2025-09-02 05:59:39.708617500 INFO     registered ourselves on D-Bus as com.victronenergy.evcharger.evc_HQ2326WG34D
```

And
```
2025-08-24 12:44:23.103840500 INFO     [tcp:192.168.1.215:502:1] Device failed: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.215:502)]
2025-08-24 12:44:43.029816500 INFO     [tcp:192.168.1.215:502:1] Found EV charger: Victron Energy AC22E
2025-08-24 12:44:43.229743500 INFO     registered ourselves on D-Bus as com.victronenergy.evcharger.evc_HQ2301V6NMA
2025-08-24 12:45:37.559973500 ERROR    Watchdog timeout
2025-08-24 12:45:37.560357500 Thread 0xb3dff400 (most recent call first):
2025-08-24 12:45:37.560363500   File "/opt/victronenergy/dbus-modbus-client/mdns.py", line 40 in recv
2025-08-24 12:45:37.560366500   File "/opt/victronenergy/dbus-modbus-client/mdns.py", line 114 in run
2025-08-24 12:45:37.560369500   File "/usr/lib/python3.12/threading.py", line 1012 in run
2025-08-24 12:45:37.560372500   File "/usr/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
2025-08-24 12:45:37.560375500   File "/usr/lib/python3.12/threading.py", line 1032 in _bootstrap
2025-08-24 12:45:37.560442500
2025-08-24 12:45:37.560444500 Current thread 0xb51b5400 (most recent call first):
2025-08-24 12:45:37.560446500   File "/opt/victronenergy/dbus-modbus-client/watchdog.py", line 21 in run
2025-08-24 12:45:37.560449500   File "/usr/lib/python3.12/threading.py", line 1012 in run
2025-08-24 12:45:37.560452500   File "/usr/lib/python3.12/threading.py", line 1075 in _bootstrap_inner
2025-08-24 12:45:37.560485500   File "/usr/lib/python3.12/threading.py", line 1032 in _bootstrap
2025-08-24 12:45:37.560514500
2025-08-24 12:45:37.560515500 Thread 0xb694b020 (most recent call first):
2025-08-24 12:45:37.560517500   File "/usr/lib/python3.12/site-packages/gi/overrides/GLib.py", line 497 in run
2025-08-24 12:45:37.560545500   File "/opt/victronenergy/dbus-modbus-client/dbus-modbus-client.py", line 452 in main
2025-08-24 12:45:37.560615500   File "/opt/victronenergy/dbus-modbus-client/dbus-modbus-client.py", line 455 in <module>
2025-08-24 12:45:37.594652500 *** starting dbus-modbus-client ***
```

Possible cause is that at the moment there [doesn't seem to be a retry when the read fails](https://github.com/victronenergy/dbus-modbus-client/blob/5ee86c931d31801e5ad5ed7ff4e2535882920cdd/device.py#L117). .And I also don't see an upper limit to the timeout [here](https://github.com/victronenergy/dbus-modbus-client/blob/5ee86c931d31801e5ad5ed7ff4e2535882920cdd/device.py#L508), probably causing the watchdog to kick in.

mansr, 2025-09-02:
The first log extract shows repeated loss of connection and failure to reconnect immediately being correctly reported. This is most likely caused by a poor network and is not indicative of a bug in dbus-modbus-client.

There are retries within pymodbus, so there is no need to do it explicitly in application code. Although there is no explicit limit to the timeout, it is calculated from observed response times.

The watchdog stack trace suggests a known problem (probably a deadlock) somewhere inside glib, which is why the watchdog was added in the first place.

dirkjanfaber, 2025-09-03:
I should have included a critical technical detail in my initial report. The error message I'm also seeing is: `No response received, expected at least 8 bytes (0 received)`. From another system (the user that suffers a lot from it stopped his remote support).:
```
2025-08-21 12:43:39.371416500 INFO     [tcp:192.168.8.21:502:1] Found EV charger: Victron Energy AC22E
2025-08-21 12:43:39.741226500 INFO     registered ourselves on D-Bus as com.victronenergy.evcharger.evc_HQ2236DNZ2U
2025-08-21 16:30:07.647385500 INFO     [tcp:192.168.8.21:502:1] Device failed: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.8.21:502)]
2025-08-21 16:30:19.302306500 INFO     [tcp:192.168.8.21:502:1] Found EV charger: Victron Energy AC22E
2025-08-21 16:30:20.735121500 ERROR    [tcp:192.168.8.21:502:1] Error reading register 0x13a3: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2025-08-21 16:32:33.863850500 INFO     [tcp:192.168.8.21:502:1] Found EV charger: Victron Energy AC22E
2025-08-21 16:32:34.107408500 INFO     registered ourselves on D-Bus as com.victronenergy.evcharger.evc_HQ2236DNZ2U
2025-08-21 18:32:52.810334500 INFO     [tcp:192.168.8.21:502:1] Device failed: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.8.21:502)]
```

This indicates a complete communication timeout where no data is received at all. Looking at the current code, when `read_modbus()` returns an error (regardless of whether pymodbus attempted internal retries), the application immediately raises an exception without any application-level retry attempt, causing the device to go offline for 10+ seconds.

Regarding network conditions: you're absolutely right that the network infrastructure may be the causing factor. However, regardless of the root cause, temporary communication interruptions are common in real-world deployments. The goal should be to improve resilience to these transient issues.

dirkjanfaber, 2025-09-10:
As the switch from modbus to mqtt for the EVCS is around the corner, it is probably worth waiting for that to be finished before spending a lot of energy on this.