victronenergy/venus #1526: Add user option to disable default gateway for Ethernet and/or Internet connection

mpvader, 2025-09-05:
https://victrondevelopment.slack.com/archives/C7C1ZU6UD/p1755873368915159

One toggle switch for Ethernet, one toggle switch for Wifi, both default enabled.

mansr, 2025-09-27:
The branches below add two new items to `com.victronenergy.platform`:
- `Network/Ethernet/GatewayEnabled`
- `Network/Wifi/GatewayEnabled`

These control whether each connection is allowed to provide a gateway. They default to 1/true, keeping existing behaviour unchanged. If set to 0/false, the corresponding network is prevented from being used for internet access even if a gateway is provided by DHCP or manually configured. If both are disabled, only a 3G/4G modem (if present) will be used.

https://github.com/victronenergy/meta-victronenergy/commits/mans/gateway
https://github.com/victronenergy/venus-platform/commits/mans/gateway

The GUI should now be updated to expose these controls. I would suggest that the gateway entry be disabled if these options are turned off in order to avoid confusion.

mansr, 2025-11-25:
Maybe move the Ethernet option either before or after all the IP address settings.